kiibohd/controller #358: K-Type not working on macOS

pottar, 2019-06-05:
* **Device**:
 * K-Type (Massdrop November 2017)

* **Firmware Origin**:
Original factory flash

* **OS**:
macOS Mojave 10.14.5

* **Version**:
669

* **Reproduction Steps**:
I connect the keyboard to the computer.

* **Resulting Issue**:
The keyboard does not respond to any key presses. If I try connecting and/or disconnecting the keyboard while monitoring log output, i get this:

```
$ log stream | grep -i usb
2019-06-05 16:52:40.508179+0000 0x74       Default     0x0                  0      0    kernel: Keyboard - K-Type PixelMap pjrcUSB full::handlePowerDomainDidChangeTo not in power tree
2019-06-05 16:52:42.569916+0000 0xcce1d    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 042393.434916 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-06-05 16:52:46.657341+0000 0x74       Default     0x0                  0      0    kernel: Keyboard - K-Type PixelMap pjrcUSB full::handlePowerDomainDidChangeTo not in power tree
2019-06-05 16:52:56.378711+0000 0xcceb7    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 042407.243548 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002c0)
```

haata, 2019-06-05:
Interesting.

Would you mind flashing the firmware on it to see if the problem has been fixed? (a lot of issues have been resolved since the keyboard shipped).
https://kiibohd.github.io/wiki/#/Quickstart

pottar, 2019-06-13:
Hi!

No, I am not using a C to C cable. I've been trying to connect the keyboard in 2 different ways: using a USB C Apple multiport and also using a Samsung USB C to USB A adapter.

After flashing the keyboard from the configurator the same problem still persists on macOS. However, now the keyboard is also kinda wonky on windows; it will take more time to be detected and sometimes a notification will appear saying the device has an error.

haata, 2019-06-18:
Have you tried using the other USB port as well?
It's possible there could be an issue with the cable, or the connector has gotten dirty.
(the symptom looks like USB errors)

The connector can be cleaned with some isopropyl alcohol and a syringe.

pottar, 2019-06-18:
I just tried with 3 different cables (2 were A to C and one was C to C) and it still does not work. I found another couple syslog messages which might be useful.

```
kernel: Keyboard - K-Type PixelMap pjrcUSB full::handlePowerDomainWillChangeTo not in power tree
kernel: Keyboard - K-Type PixelMap pjrcUSB full::handlePowerDomainDidChangeTo not in power tree
kernel: Keyboard - K-Type PixelMap USB::handlePowerDomainDidChangeTo not in power tree
```

It seems to be an issue with either power delivery on macOS or on the power configuration in the keyboard controller, or both.

kubedzero, 2019-07-07:
Just want to say I had this issue too (the weird power tree/log issues) with my computer, although it only happens maybe 10% of the time and with one specific port on my 2018 13" Macbook pro touchbar. Chances of success are far higher when I use the other ports, from what I've seen. I tried re-flashing it with a lower global brightness (thinking that the amperage it was pulling at startup was too much) but that didn't seem to increase the success rate. I'll update here if I find anything concrete!

pottar, 2019-07-16:
Hey, I just saw this, sorry for the delay. I had to switch to QMK since I needed the keyboard working ASAP and that didn't have this problem, but I do wanna switch back as I miss RGB functionality...

Now that you said that about global lighting, can you try disabling the LEDs and testing it? I'd do that but I don't have much time these coming days. If it doesn't help then it might be the code that handles power stuff with the OS and not the actual power requirements of the keyboard. Since you brought that up maybe it's working now because it is not powering up the LEDs as QMK has no support for in this keyboard.

dcchambers, 2019-11-14:
+1 similar issue. Very sporadic on my 2019 MBP with Mac OS 10.14.6. Using the USBC-to-A cable that came with my K-type, plugged into the official Apple USB-A to USB-C adaptor.