kiibohd/controller #360: Media "Stop" key picked up as "Cancel" on Linux

tomiga, 2019-07-13:
* **Device**:
Kira

* **Firmware Origin**:
Configurator 1.0.1

* **OS**:
Debian 10 "Buster" on x86_64

* **Version**:
bcdDevice shows 4.8b

* **Reproduction Steps**:
Press any key bound to the "Stop" key.

* **Resulting Issue**:
Key is picked up as "Cancel", not allowing it to be picked up as the media Stop key on Linux.

Relevant xev output:
```
KeyPress event, serial 82, synthetic NO, window 0x3e00001,
    root 0x6cb, subw 0x0, time 7813744, (373,717), root:(467,875),
    state 0x0, keycode 136 (keysym 0xff69, Cancel), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 82, synthetic NO, window 0x3e00001,
    root 0x6cb, subw 0x0, time 7813924, (373,717), root:(467,875),
    state 0x0, keycode 136 (keysym 0xff69, Cancel), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```
I can confirm I don't have this issue with other keyboards on this system.

haata, 2019-07-18:
Interesting. Do you have an example from a different keyboard?
(I'm rather interested in what other keyboards set it to as this is usually pretty straightforward to fix).

tomiga, 2019-07-20:
Sorry about the slow response. Here's what an old Razer BlackWidow keyboard of mine outputs where the stop key does the desired action:
```
FocusOut event, serial 78, synthetic NO, window 0x4200001,
    mode NotifyGrab, detail NotifyAncestor

FocusOut event, serial 78, synthetic NO, window 0x4200001,
    mode NotifyUngrab, detail NotifyPointer

FocusIn event, serial 78, synthetic NO, window 0x4200001,
    mode NotifyUngrab, detail NotifyAncestor

KeymapNotify event, serial 78, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

MappingNotify event, serial 78, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248
```