kiibohd/controller #361: Kira preventing Linux CSGO from launching, held inputs in other games.

Aigis, 2019-07-19:
- **Device**:
Kira
- **Firmware**:
Current
- **OS**:
Manjaro or Arch
- **Version**:
Current (As of time of posting, 7/19/2019)
- **Reproduction Steps**:
Have Kira plugged in while launching game through Steam
- **Resulting Issue**:
Game will either hang after a brief black screen (CSGO), load and run incredibly slow (DOTA2), will launch but will have what appears to be an up input held on menus (SoulCalibur VI and Tekken 7) or in some cases fail to accept inputs. Launching CSGO without the keyboard plugged in will yield a boot, and re-plugging in the keyboard will allow the keyboard to function normally, but not before reconnecting it.

- **Additional info**:
For the sake of thoroughness, I checked to see if there was any key chatter on the board through Switchhitter on Windows, no signs of issues. I have tried every USB port on my computer, same yield. I have tried an old Dell USB keyboard, a GH60, a KBD75 by KBDFans, an XD84 by KPRepublic as well as an XK87 by KPRepublic. This issue only happens with the Kira. To test more I have tested this against XFCE, KDE as well as Cinnamon. The issue happens identically across all of this. I can't say I know enough about how xorg works to tell you too much without someone else helping to tell me what to look for, but xev doesn't report any keys held down.

haata, 2019-07-19:
Interesting.

Can you get a little more information for me?

(this will enter the keyboard cli, and run some status commands)
```bash
sudo screen /dev/ttyACM0
outputDebug
```

If nothing shows up, press a single key and a bit field should show up.
(I'm looking for whether something on the keyboard seems to think it's held).

Switching to 6KRO mode might help a bit but I'm not certain at this point.

Aigis, 2019-07-20:
Sure thing. So after "outputDebug" I get this line returned
: DEBUG - NKRO USB: NKRO 00 000000000000000000000000000000000000000000 000000000000
Which appears to be the line that's returned after every keystroke after.

haata, 2019-07-20:
Very interesting, sounds like I'll need to purchase some steam games to test this out.

(your keyboard is behaving correctly, so it may be in how the USB HID descriptor is being interpreted).

Aigis, 2019-07-21:
If it's any consolation, CSGO is free to play. Is there any other data I can gather you, either from the Kira or from another keyboard?

haata, 2019-07-21:
Hmm, maybe some xev data on the Linux side might be useful.
I'm also curious about what happens when you press something like CapsLock in game (after you got the keyboard to work correctly).

Also, does it help if the keyboard is in 6KRO mode when you enter the game?

Aigis, 2019-07-21:
6KRO (just to make sure I'm doing this right, rctrl+rshift + F2 will switch to 6KRO, or F1 will switch between NRKO and 6KRO, correct?). Trying the swap, 6KRO or NKRO settings through the keyboard do not change the behavior.

CapsLock does not change the behavior either, everything appears to be working as intended once the keyboard is plugged in post launching the game. 

Here is a xev output of my XD87 launching and running CSGO for a minute or so - https://gist.github.com/Aigis/fa499442451c5600e7f3a8019b7a0905

Aigis, 2019-10-09:
Is there any update on this? Alternatively is there any way to just run QMK on the keyboard? I'd like to actually use the keyboard I paid a premium on...

haata, 2019-10-09:
Sorry about the delay. I just tried CS GO (Arch Linux, installed everything today) and I didn't have any issues playing. Do you know if there are any logs for CS:GO?

haata, 2019-10-09:
(I am using slightly updated firmware on the keyboard, but I don't think I've changed anything on the USB side).

Aigis, 2019-10-11:
Sorry for the delay on the response (some family things cropped up). CSGO and DOTA2 now appear to boot properly (I haven't kept up with updates over the last few months, so I'm assuming that Valve patched something). Launching back up Tekken 7 results in character still holding up and repeatedly jumping, (tested with both the default Proton 3.16 and the newest 4.11), and SoulCalibur VI still just holds up on the menu. 
This behavior made me think why do the menus in T7 work but not in SCVI? Testing with an Xbox One controller shows me that T7 accepts inputs from only the DPad on menus, but works with the analog stick in game, whereas SCVI accepts menu movement from the analog stick as well. So while I wasn't able to give you a crash log for CSGO (As it appears to be working now), I hope that this extra info about the Proton games helps

haata, 2019-10-11:
Interesting. Have you tried using 6KRO mode in Tekken 7 or Soulcalibur VI?

Aigis, 2019-10-11:
Same issue in NKRO and 6KRO