kiibohd/controller #362: Dead key after switch swap

rynodigital, 2019-07-20:
* **Device**:
Kira

* **Firmware Origin**:
Web Configurator

* **OS**:
macOS Mojave
10.14.5

I'm sorry if this isn't the right place to post about this, but I've recently tried swapping out switches on my Kira and my 6 key on row 2 stopped working.  The LED functions completely fine but the key press itself does not register anymore.

Is there anything I can do to fix this without having to open up the board?
[kira_json.txt](https://github.com/kiibohd/controller/files/3413068/kira_json.txt)

haata, 2019-07-22:
If the 6 key is the key you swapped, it could possibly be that one of the switch pins are bent. You'll need to remove it and re-insert the switch.

Another possibility is that the pcb was damaged while inserting the switch (sometimes a bent pin, or a switch pin with solder on it can push out the hotswap socket).

It's a bit easier to tell with a picture of the pcb and underside of the switch (just remove the switch).

rynodigital, 2019-07-22:
I saw someone had a similar issue on Reddit, and it turned out to be that the hot swap circut came loose.  I'm hoping that's the problem!