kiibohd/controller #370: Kira: Layer visualizations can "stick" even when the layer has been exited

notJerl, 2019-10-27:
* **Device**:
Kono Kira

* **Firmware Origin**:
Kiibohd Configurator

* **OS**:
Windows 7

* **Reproduction Steps**:
1. Define a macro from the base layer to lock another layer.  Let's say LAlt+LShift+NumLk to lock layer 3, since in my case I'm using this layer to replace Windows' MouseKeys.
2. Define a key *that's part of the above macro* on this new layer to activate (not lock or latch) a higher layer. Let's say Shift to activate layer 4.  As an explanation, I'm using the second layer to replace the fast mouse movement keybinds on the lower layer with slow mouse movement keybinds, and since I'm using Z, X, and C for mouse buttons, shift makes sense as a modifier to slow movement.
3. Define the following layer highlights on the base layer to make the LEDs for the keys on layer 3 to flash:
`Layer[3] : fade_layer_highlight(3);`
(I didn't define it for layer 4 because layer 4 doesn't have any bindings that aren't already on layer 3, and there isn't a defined way to access layer 4 outside of layer 3)
3. Use the base layer macro to lock layer 3.
4. Use the bound key to access layer 4 on top of layer 3.
5. While still holding that key, use the base layer macro to unlock layer 3.
* **Resulting Issue**:
Layers 3 and 4 are deactivated as expected.  However, the LED's under the keys that were part of layer 4 are still flashing, even though layer 4 is not active.  The only way to make them stop flashing is to activate layer 4 again. the LED TOG/LED OFF keys will turn them off with the rest of the visualization, but they start flashing again as soon as LED TOG/LED ON is pressed.  V:RESET does not stop the LEDs flashing.  Selecting a different visualization does not stop the LEDs flashing.

haata, 2019-11-03:
Interesting. Some possible things that are happening:
1. The Layer change signal is not getting emitted
2. A bug in the layer change signal that's looking at incorrect status
3. Signal from layer is not being used (less likely)
4. Signal from layer not assigned using KLL (unlikely, you've already mentioned doing this above)

This kind of issue is a bit easier to debug with a debug log.

https://kiibohd.github.io/wiki/#/Debugging
Use the `capDebug` command and run through your key combination, then post the results here.

notJerl, 2019-11-05:
```
capDebug
INFO - Capability Debug Mode: 1
: DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 40
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerLock(layerIndex) -> 3
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerLock(layerIndex) -> 3
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Un-safe: Pixel_FadeLayerHighlight_capability(layer)
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerLock(layerIndex) -> 3
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerLock(layerIndex) -> 3
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerLock(layerIndex) -> 3
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerLock(layerIndex) -> 3
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerLock(layerIndex) -> 3
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerLock(layerIndex) -> 3
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 230
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
[... 396 lines cut ...]
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 83
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Output_usbCodeSend(usbCode) -> 229
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
DEBUG - Safe: Layer_layerShift(layerIndex) -> 4
```

This is for pressing and releasing RAlt+RShift+NumLock, then pressing and holding LShift, then while LShift is still held pressing and releasing RAlt+RShift+NumLock, and then letting go of LShift.  I have no idea if all those repeated lines are normal or not.