kiibohd/controller #371: Hexgears Gemini not working on macOS

verydanny, 2019-11-12:
* **Device**:
  - Hexgears Gemini prebuy (stock layout)

* **Firmware Origin**:
  - Hexgears Configurator (updated to latest) and reflashed

* **OS**:
  - macOS 10.14.6 (18G1012)
  - iMac Pro 2018

* **Version**:
  - Below is screenshot of info:

[image]

**Reproduction Steps**:

1. Plug in keyboard
2. Keyboard LEDs start flashing on and off
3. Keyboard either starts working, or keeps flashing indefinitely.

* **Resulting Issue**:

Keyboard does not function as intended. If it does work, it does so after anywhere from 15s to until I plug it out then back in again. This is the a USB power issue. It doesn't work on powered hubs, or hubs directly on motherboard. I cleaned all USB-C ports with alcohol. I tried many different things.

It only works on 1 of the macs I own, but only after about 15s of the same glitchy LED flashing.

Here is a log of usb messages while it attempts to "connect" the keyboard, the keyboard is connected to the HS02 (High Speed 02 port):

```bash
$ log stream | grep -i usb
2019-11-12 02:37:11.572729+0000 0xfa8fb    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 038961.432895 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-11-12 02:37:13.793508+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 038963.653619 HS02@14200000: AppleUSBHostPort::forcePowerGated: kPowerStateSuspended completed with 0xe00002d6 after 1411ms
2019-11-12 02:37:14.119442+0000 0x80a      Default     0x0                  0      0    kernel: (AppleUSBXHCI) 038963.979528 HS02@14200000: AppleUSB20XHCIPort::resume: unexpected link state (PORTSC 0x0e4202a0)
2019-11-12 02:37:14.219777+0000 0x200      Default     0x0                  0      0    kernel: (IOUSBFamily) 038964.079885 Keyboard - Gemini_Dusk_Dawn PixelMap USB@14200000: AppleUSBDevice::start: failed to attach or start _deviceClient
2019-11-12 02:37:14.219844+0000 0x200      Default     0x0                  0      0    kernel: (IOUSBFamily) 038964.079952 AppleUSBLegacyRoot@(null): AppleUSBLegacyRoot::usbServiceCallGated: failed to attach or start legacy service
2019-11-12 02:37:17.589901+0000 0xfa969    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 038967.449939 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-11-12 02:37:22.197299+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 038972.057237 HS02@14200000: AppleUSBHostPort::forcePowerGated: kPowerStateSuspended completed with 0xe00002d6 after 1410ms
2019-11-12 02:37:22.298840+0000 0x74       Default     0x0                  0      0    kernel: Keyboard - Gemini_Dusk_Dawn PixelMap USB::handlePowerDomainDidChangeTo not in power tree
2019-11-12 02:37:28.001066+0000 0xfaa37    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 038977.860883 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-11-11 21:37:29.660037-0500 0x2fb      Default     0x0                  99     0    bluetoothd: (IOBluetooth) USBProductID 0x828D	USBVendorID 0x5AC
2019-11-11 21:37:29.660532-0500 0x2fb      Default     0x0                  99     0    bluetoothd: (IOBluetooth) USBProductID 0x828D	USBVendorID 0x5AC
2019-11-12 02:37:30.220453+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 038980.080221 HS02@14200000: AppleUSBHostPort::forcePowerGated: kPowerStateSuspended completed with 0xe00002d6 after 1410ms
2019-11-12 02:37:36.118427+0000 0xfac1a    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 038985.978070 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-11-12 02:37:38.408831+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 038988.268424 IOUSBHostDevice@14200000: IOUSBHostDevice::start: failed to get device descriptor
2019-11-12 02:37:38.408965+0000 0x131      Default     0x0                  0      0    kernel: (AppleUSBXHCI) 038988.268561 HS02@14200000: AppleUSBXHCIPort::resetAndCreateDevice: failed to start device
2019-11-12 02:37:38.408973+0000 0x131      Default     0x0                  0      0    kernel: (AppleUSBXHCI) 038988.268570 HS02@14200000: AppleUSBXHCIPort::resetAndCreateDevice: failed to create device (0xe00002e9)
2019-11-12 02:37:38.409031+0000 0x80a      Error       0x0                  0      0    kernel: (IOUSBHostFamily) AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe <private> failed (0xe00002eb)
2019-11-12 02:37:38.409123+0000 0xfac1e    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 038988.268718 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-11-12 02:37:38.409128+0000 0xfa904    Error       0x0                  0      0    kernel: (IOUSBHostFamily) AppleUSBHostRequestCompleter::clearPipeStallThreadCall: clearStall for pipe <private> failed (0xe00002eb)
2019-11-12 02:37:38.409961+0000 0x74       Default     0x0                  0      0    kernel: Keyboard - Gemini_Dusk_Dawn PixelMap USB::handlePowerDomainDidChangeTo not in power tree
2019-11-12 02:37:41.566130+0000 0x74       Default     0x0                  0      0    kernel: Keyboard - Gemini_Dusk_Dawn PixelMap USB::handlePowerDomainDidChangeTo not in power tree
2019-11-12 02:37:43.957257+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 038993.816726 HS02@14200000: AppleUSBHostPort::forcePowerGated: kPowerStateSuspended completed with 0xe00002d6 after 1410ms
2019-11-12 02:37:46.419346+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 038996.278793 IOUSBHostDevice@14200000: IOUSBHostDevice::start: failed to get device descriptor
2019-11-12 02:37:46.419500+0000 0x131      Default     0x0                  0      0    kernel: (AppleUSBXHCI) 038996.278947 HS02@14200000: AppleUSBXHCIPort::resetAndCreateDevice: failed to start device
2019-11-12 02:37:46.419521+0000 0x131      Default     0x0                  0      0    kernel: (AppleUSBXHCI) 038996.278971 HS02@14200000: AppleUSBXHCIPort::resetAndCreateDevice: failed to create device (0xe00002e9)
2019-11-12 02:37:46.419643+0000 0xfac56    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 038996.279089 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-11-12 02:37:49.656235+0000 0xfacad    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 038999.515656 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-11-12 02:37:51.878429+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 039001.737827 HS02@14200000: AppleUSBHostPort::forcePowerGated: kPowerStateSuspended completed with 0xe00002d6 after 1412ms
2019-11-12 02:37:54.271995+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 039004.131357 HS02@14200000: AppleUSBHostPort::forcePowerGated: kPowerStateSuspended completed with 0x00000000 after 1412ms
2019-11-12 02:37:57.671337+0000 0xfaccb    Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 039007.530667 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::ConfigureDevice: unable to set a configuration (0xe00002ed)
2019-11-12 02:37:59.891798+0000 0x131      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 039009.751078 HS02@14200000: AppleUSBHostPort::forcePowerGated: kPowerStateSuspended completed with 0xe00002d6 after 1412ms
2019-11-12 02:38:00.228617+0000 0xfa99b    Default     0x0                  0      0    kernel: (AppleUSBXHCI) 039010.087905 HS02@14200000: AppleUSB20XHCIPort::resume: timeout waiting for U0
2019-11-12 02:38:00.228640+0000 0xfa99b    Default     0x0                  0      0    kernel: (AppleUSBXHCI) 039010.087931 HS02@14200000: AppleUSB20XHCIPort::resume: unexpected non-U0 link state (PORTSC 0x0e4202a0)
[... cut ...]
2019-11-12 02:39:00.366096+0000 0xfa99b    Default     0x0                  0      0    kernel: (AppleUSBXHCI) 039070.224206 HS02@14200000: AppleUSB20XHCIPort::resume: unexpected link state (PORTSC 0x0e4202a0)
2019-11-12 02:39:00.468553+0000 0xfa99b    Default     0x0                  0      0    kernel: (IOUSBFamily) 039070.326664 Keyboard - Gemini_Dusk_Dawn PixelMap USB@14200000: AppleUSBDevice::start: failed to attach or start _deviceClient
2019-11-12 02:39:00.468618+0000 0xfa99b    Default     0x0                  0      0    kernel: (IOUSBFamily) 039070.326730 AppleUSBLegacyRoot@(null): AppleUSBLegacyRoot::usbServiceCallGated: failed to attach or start legacy service
```

verydanny, 2019-11-13:
**Some more information:**

It seems that when the keyboard is "attempting"  to reconnect (it keeps detaching itself), it has something to do with figuring out the `IOPowerManagement`. No other device I have has issues with power. I power a small display through USB-C on my iMac. 

I am using `IORegistryExplorer` from Apple's Additional Tools for Xcode 11 (https://developer.apple.com/download/more/) to get this info.

**All attempted connections:**

[image]

I noticed that some of the connections have the `IOPowerManagement` property, but some do not:

**Connection Attempt 1 (`IOPowerManagement` present):**

[image]

**Connection Attempt 2 (`IOPowerManagement` missing):**

[image]

**Connection Attempt 3 (`IOPowerManagement` missing):**

[image]

**Connection Attempt 4 (`IOPowerManagement` missing):**

[image]

**Connection Attempt 5  (`IOPowerManagement` present):**

[image]

al45tair, 2019-11-13:
FWIW, it works fine on both my iMac and my MacBook Pro here. Well, modulo the other issues with the configurator software and so on.

If I had to guess, you've got a dicky cable somewhere; try changing to the other port, and try switching to a different USB-C cable. Also, try wiggling the cable.

verydanny, 2019-11-13:
@al45tair 

I've tried different cables and different ports, cleaning the ports, wiggling. It just doesn't wanna work.

al45tair, 2019-11-13:
Did you flash the firmware, or is this with the stock firmware? (Stock firmware works OK here; if you try to generate firmware with the configurator, it may or may not work depending on what you try to do.)

al45tair, 2019-11-13:
If you've flashed it and got it into this state, try clicking the "Standard" option on the layout menu, and use a paperclip to put the board into flash mode before clicking the Flash option.

verydanny, 2019-11-13:
I've tried putting it into "Standard". I tested it stock, standard, etc. 

It hasn't been doing anything but flashing on and off.

colingourlay, 2019-11-19:
Getting exactly the same issue here on stock firmware and configuration, verified by running the same debug tools. My keyboard was fine for  the first few days, then today I tried enabling some different lighting modes (using the 2nd layer FN keys) and it's started freaking out as described above.

colingourlay, 2020-01-27:
Hey folks, I've just upgraded Mac OS to Catalina (10.15.2) and this issue has gone away entirely. No problems connecting my Hexgears Gemini Dusk; no problems entering/exiting flash mode or flashing a custom configuration. I've no idea what Apple's changed between major versions, but they seem to have fixed it!

haata, 2020-01-27:
Yay! (Catalina has been pretty rocky, so this doesn't surprise me)