kiibohd/controller #372: Gemini Layer 2 - F3 - C:UNDER - Doesn't change underlighting

Laments, 2019-11-13:
* **Device**:
Gemini Dusk (Nov 2019) 

* **Firmware Origin**:
From Hexgears Configurator v1.0.2-r2

* **OS**:
Windows 10, v.1903

* **Version**:
0x048B

* **Reproduction Steps**:
1. Open HexGears Configurator, select Gemini Dusk
2. Set keyboard to FLASH mode
3. FLASH keyboard
4. Press Layer 2 F3 a few times.

* **Resulting Issue**:
Before flashing Gemini Dusk, in Layer 2, the F3 key (mapped to C:UNDER) changes underlighting. After flashing, Layer 2 F3 will light up random LEDs under the keyboard, and not change underlighting anymore.

al45tair, 2019-11-13:
FYI, re-flashing with firmware built from this git repo fixes this.

smasher816, 2019-11-16:
The default layout in the configurator has been updated to fix this.

Please select `Layouts -> Standard` at the top right to load the corrected layout.

closed 2019-11-16 by smasher816