kiibohd/controller #375: K-Type does not work reliably with Mac.

dcchambers, 2019-11-14:
* **Device**:
  * K-Type (Massdrop November 2017)

* **Firmware Origin**:
  * Configurator, flashed ~June 2019.

* **OS**:
  * MacOS 10.14.6 
  * 2019 13" MBP

* **Version**:
  * ?

* **Reproduction Steps**:
  * To reproduce this issue, simply plug in K-Type keyboard to a MBP. Does not register any keypresses in MacOS. Lights flash on and off randomly.
  * I am using the USB-C-to-USB-A cable that was included with the K-Type keyboard. I am plugging this into the official Apple USB-A to USB-C dongle, and plugging that into the laptop (which only has USB-C ports).
  * **Sometimes** it will work. Most of the time (9/10) it does not. It's random. I have tried every USB-C port on the laptop and both USB-C ports on the keyboard. I have not tried a direct USB-C to USB-C connection as I don't have a spare USB-C cable right now.

* **Resulting Issue**:
  * MacOS does not recognize keypresses from the keyboard. The backlights on the keyboard flash on and off randomly.

* **Output from USB log**
```
10:39:20 ~ ❱❱ log stream | grep -i usb                                                                                                                                                                              
2019-11-14 16:39:36.091362+0000 0x33f0738  Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 1236080.086444 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::Configu
reDevice: unable to set a configuration (0xe00002c0)                                                                                                                                                                
2019-11-14 16:39:40.129625+0000 0x33f0773  Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 1236084.124695 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::Configu
reDevice: unable to set a configuration (0xe00002c0)                                                                                                                                                                
2019-11-14 16:39:44.168019+0000 0x33f0799  Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 1236088.163085 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::Configu
reDevice: unable to set a configuration (0xe00002c0)                                                                                                                                                                
2019-11-14 16:39:54.182289+0000 0x33f0800  Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 1236098.177324 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::Configu
reDevice: unable to set a configuration (0xe00002c0)                                                                                                                                                                
2019-11-14 16:40:18.265709+0000 0x33f0ccb  Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 1236122.260667 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::Configu
reDevice: unable to set a configuration (0xe00002c0) 
2019-11-14 16:40:40.236373+0000 0x33f0e04  Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 1236144.231270 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::Con
figureDevice: unable to set a configuration (0xe00002c0)
2019-11-14 16:40:46.220190+0000 0x33f0e41  Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 1236150.215068 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::Configu
reDevice: unable to set a configuration (0xe00002c0)                                                                                                                                                                
2019-11-14 10:41:18.588300-0600 0x33f16c5  Default     0x0                  35366  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-14 16:41:18.588060+0000 0x33f16c5  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236182.582852 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                                                                                                                                                                                        
2019-11-14 16:41:18.588852+0000 0x33f16c5  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236182.583645 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control
2019-11-14 16:41:18.588852+0000 0x33f16c5  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236182.583645 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                                                                                                                                                                                        
2019-11-14 10:41:18.588968-0600 0x33f16c5  Default     0x0                  35366  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-14 16:41:18.589533+0000 0x33f16c5  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236182.584329 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                         
2019-11-14 10:41:18.589618-0600 0x33f16c5  Default     0x0                  35366  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-14 16:41:18.591382+0000 0x33f16c5  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236182.586177 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                         
2019-11-14 10:41:18.591456-0600 0x33f16c5  Default     0x0                  35366  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-14 10:41:18.591951-0600 0x33f16c5  Default     0x0                  35366  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-14 16:41:18.591889+0000 0x33f16c5  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236182.586685 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                         
2019-11-14 16:41:20.294519+0000 0x33f1797  Default     0x0                  0      0    kernel: (AppleUSBHostCompositeDevice) 1236184.289307 AppleUSBCDCCompositeDevice@(null): AppleUSBHostCompositeDevice::Configu
reDevice: unable to set a configuration (0xe00002c0) 
2019-11-14 16:41:44.465221+0000 0x33f222f  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236208.459938 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                         
2019-11-14 10:41:44.465491-0600 0x33f222f  Default     0x0                  36456  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-14 16:41:44.466230+0000 0x33f222f  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236208.460946 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                         
2019-11-14 10:41:44.466443-0600 0x33f222f  Default     0x0                  36456  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-11-14 16:41:44.467023+0000 0x33f222f  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236208.461742 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                         
2019-11-14 10:41:44.467109-0600 0x33f222f  Default     0x0                  36456  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
[... cut ...]
2019-11-14 16:41:44.470077+0000 0x33f222f  Default     0x0                  0      0    kernel: (IOUSBFamily) 1236208.464795 system_profiler@(null): IOUSBUserClientLegacy::start: missing entitlement com.apple.ibr
idge.control                                         
2019-11-14 10:41:44.470162-0600 0x33f222f  Default     0x0                  36456  0    system_profiler: (SPUSBReporter) SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
^C                                                   
10:41:50 ~ ❱❱                                                        
```

dcchambers, 2019-11-15:
I tried a different USB-C to USB-A cable that I had (From Google) - it did not work either.

I tried a USB-C to USB-C cable (from Apple) directly from the laptop to the keyboard, and that did not work either.

colingourlay, 2019-11-19:
Sounds like https://github.com/kiibohd/controller/issues/371 (which is affecting me)

dcchambers, 2019-11-19:
Also similar to #358 - seems like this is a USB power/hardware issue. Either with MacOS or with the controller on the keyboard.

pcolusso, 2019-11-21:
Recently begun experiencing this issue after I flashed a new map onto the device. Unit was working fine prior, and had been flashed back in 2018.

Keyboard appears to restart constantly, and after about ten minutes, begins working normally again.

Have tried an AmazonBasics USB-A to C cable and a generic USB-C cable to my dock, an Apple USB-C to USB-C directly into the machine. System is a 2017 MacBook Pro.

intafon, 2019-11-21:
I had the same issue today -- I'm guessing it has to do with some USB issue on the mac that was introduced recently. It looks like restarting the mac fixed the issue for me (at least until I tried to reconfigure in kiibohd config again, at which point it freaked out again and I had to restart again). The flashed firmware actually worked, but no input was coming from the keyboard until I restarted...

colingourlay, 2020-01-27:
Hey folks, I've just upgraded Mac OS to Catalina (10.15.2) and this issue has gone away entirely. No problems connecting my Hexgears Gemini Dusk; no problems entering/exiting flash mode or flashing a custom configuration. I've no idea what Apple's changed between major versions, but they seem to have fixed it!

intafon, 2020-01-27:
I too upgraded recently to Catalina. I hadn't tried running kiibohd config again until just now, and it worked for me as well...

haata, 2020-01-27:
Interesting! Thanks for the update. Please let me know if there are any more issues.
(I'm working on a bunch of USB things for HID-IO features)

dcchambers, 2020-01-28:
Good to know that Catalina fixes it - but unfortunately I am unable to upgrade my work laptop right now and will be on 10.14 for the forseable future...

intafon, 2020-01-28:
Also sounds like your case was more exaggerated than mine. In my case, restarting the computer would get it talking to the keyboard fine again...

dcchambers, 2020-02-10:
@intafon I can confirm that restarting mac fixes the issues and the keyboard works as expected. However any time you flash a new firmware to the keyboard you'll have to restart your computer again.