kiibohd/controller #379: K-Type enters flashing mode when using Ultimaker Cura

gabfv, 2019-12-20:
* **Device**:
K-Type (Massdrop November 2017)

* **Firmware Origin**:
Web configurator

* **OS**:
Windows 10 Pro

* **Version**:

> Information for device Keyboard - K-Type PixelMap USB (VID=0x308F PID=0x0011):
> 
> Connection Information:
> ------------------------------
> Device current bus speed: FullSpeed
> Device supports USB 1.1 specification
> Device supports USB 2.0 specification
> Device address: 0x0004
> Current configuration value: 0x00
> Number of open pipes: 0
> 
> Device Descriptor:
> ------------------------------
> 0x12	bLength
> 0x01	bDescriptorType
> 0x0200	bcdUSB
> 0x00	bDeviceClass      
> 0x00	bDeviceSubClass   
> 0x00	bDeviceProtocol   
> 0x40	bMaxPacketSize0   (64 bytes)
> 0x308F	idVendor
> 0x0011	idProduct
> 0x048B	bcdDevice
> 0x01	iManufacturer
> 0x02	iProduct
> 0x03	iSerialNumber
> 0x01	bNumConfigurations
> 
> Configuration Descriptor:
> ------------------------------
> 0x09	bLength
> 0x02	bDescriptorType
> 0x00CF	wTotalLength   (207 bytes)
> 0x07	bNumInterfaces
> 0x01	bConfigurationValue
> 0x04	iConfiguration
> 0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
> 0xFA	bMaxPower      (500 mA)
> 
> Interface Descriptor:
> ------------------------------
> 0x09	bLength
> 0x04	bDescriptorType
> 0x00	bInterfaceNumber
> 0x00	bAlternateSetting
> 0x01	bNumEndPoints
> 0x03	bInterfaceClass      (Human Interface Device Class)
> 0x01	bInterfaceSubClass   
> 0x01	bInterfaceProtocol   
> 0x05	iInterface
> 
> HID Descriptor:
> ------------------------------
> 0x09	bLength
> 0x21	bDescriptorType
> 0x0111	bcdHID
> 0x00	bCountryCode
> 0x01	bNumDescriptors
> 0x22	bDescriptorType   (Report descriptor)
[... 248 lines cut ...]
> 0x07   0x0000  "Media Keys"
> 0x08   0x0000  "Virtual Serial Port - Status"
> 0x09   0x0000  "Virtual Serial Port - Data"
> 0x0A   0x0000  "Mouse"
> 0x0B   0x0000  "RawIO API Interface"
> 
> ------------------------------
> 
> Connection path for device: 
> USB xHCI Compliant Host Controller
> Root Hub
> Keyboard - K-Type PixelMap USB (VID=0x308F PID=0x0011) Port: 6
> 
> Running on: Windows 10 or greater (Build Version 18362)
> 
> Brought to you by TDD v2.12.0, Aug  2 2019, 16:35:05

* **Reproduction Steps**:

1. Install Ultimaker Cura (I am using version 4.4.0).
2. Launch Ultimaker Cura.
3. After a short period of time, the keyboard enters flashing mode. Pressing ESC on the keyboard does bring it out of the flashing mode, but eventually, it re-enters flashing mode.

* **Resulting Issue**:
The keyboard is in flashing mode. Pressing ESC on the keyboard does bring it out of flashing mode, but eventually, it re-enters flashing mode shortly after.

gabfv, 2019-12-20:
I've created an [issue](https://github.com/Ultimaker/Cura/issues/6833) with Cura as well, since I did not know whether the issue resides in my keyboard or in Cura.

haata, 2019-12-20:
Yeah, the K-Type has a bunch of USB endpoints. Most likely an unexpected USB request is being sent over one of the reserved endpoints which is crashing the keyboard. In some cases the K-Type will reset to flash mode when the firmware crashes. I believe later firmware generally resolves this, but possibly not for all situations.
If you can find out which endpoints are being used by Cura, then I can give more insight and possibly suggest a solution.

gabfv, 2019-12-21:
Seems like it just probes all serial ports automatically. I've disabled the plugin so this fix the issue I've encountered.

Ghostkeeper, 2019-12-23:
Cura has an open feature request regarding this: https://github.com/Ultimaker/Cura/issues/3449

gabfv, 2019-12-23:
Thanks, I'll close this issue since it lies in Cura.

closed 2019-12-23 by gabfv