kiibohd/controller #382: Errors running ergodox.bash

amriddell, 2020-01-05:
Hi,

I'm running OSX Catalina and have updated everything related to creating the Ergodox Infinity binaries.
Although the path to kll looks wrong I even get the same error when it is correct.
Also, this is just a build "out of the box".
Any thoughts? Anyone else had this kind of problem?
Thanks.
Alistair

-- Build OS Detected:
Generic
Invalid version of ctags found. Ignoring...
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ctags: illegal option -- -
usage: ctags [-BFadtuwvx] [-f tagsfile] file ...
-- Checking for kll
/Users/amr/.local/share/virtualenvs/Keyboards-JbttBO5C/bin/python3: No module named kll.__main__; 'kll' is a package and cannot be directly executed
CMake Error at Lib/CMake/kll.cmake:94 (message):
  kll compiler is not available 'python3;-m;kll': .  You may need to set
  KLL_EXECUTABLE manually.  Or check the Pipfile.
Call Stack (most recent call first):
  CMakeLists.txt:200 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/amr/ErgoDox/controller/Keyboards/darwin19.my_layout-r.gcc.ninja/CMakeFiles/CMakeOutput.log".
Error in cmake. Exiting...

amriddell, 2020-01-05:
``

amriddell, 2020-01-05:
Oops, accidentally closed this (I think) but didn't mean to.

haata, 2020-03-24:
Updated the docs a bit for this: https://github.com/kiibohd/controller/tree/master/Keyboards

Missing pipenv setup steps.

closed 2020-03-24 by haata