kiibohd/controller #383: GEMINI RGUI (possibly MENU as well) remap issue due to layout mismatch

Laments, 2020-02-01:
**Device:**
Gemini Dusk (Nov 2019)

**Firmware Origin:**
From Hexgears Configurator v1.0.2-r2

**OS:**
Windows 10, v.1903

**Version:**
0x048B

* **Reproduction Steps**:
1. Assign RGUI to any key on Base Layer
2. Flash keyboard
3. Reassign RGUI to be LOCK-1
4. Flash keyboard again
5. Observe: Menu comes up when hitting RGUI instead of LOCK layer engaging.

* **Resulting Issue**:
It's not possible to remap RGUI because the configurator and firmware KLL layouts have a mismatch.

_**Please note**_: _I believe this also may occur for MENU key as well_.

ChaosinaCan, 2020-02-01:
Related: https://github.com/kiibohd/configurator/issues/152

mathewng, 2020-10-26:
I just bought this keyboard too and I have this workaround.
Adding the two lines below into **Custom KLL** section in the hexgears configurator defines the MENU and FN keys.

```
S106 : Layer[1];
S108 : U"App";
```

You can amend the right side of the colon to any key you want to output.