kiibohd/controller #384: Support for Aurora RGB sync

JoeShoe1121, 2020-02-05:
Was wondering if it would be possible to add some sort of support for Aurora RGB sync
https://github.com/antonpup/Aurora

Im not quite familiar with Kiibohd controller, but my understanding of the aurora software is that the keyboard needs to have controls from the computer side in order to work properly.  Anyway this can be bypassed?

haata, 2020-02-05:
This will require HID-IO integration (currently working on it!), but this should be possible to support rather easily. It's a little sad that it's Windows only (from my investigation so far), but something is better than nothing :D

JoeShoe1121, 2020-04-10:
out of curiosity, any updates thus far?

haata, 2020-04-15:
Slowly. I have a few small projects to finish up before getting back into HID-IO work.