kiibohd/controller #388: Hexgears Gemini white backlighting not white

kuanweeloong, 2020-05-12:
Hi all,

I have a Hexgears Gemini Dusk which I bought from Kono in March 2020. Up till now, I not been able to get white backlighting by setting the backlight to color #FFFFFF using the hexgears configurator.

It looks like this is a known issue (See 'known bugs' section in the [quickstart guide](https://kiibohd.github.io/wiki/#/Keyboards/GeminiDawn)) - is there any chance this will get fixed? The suggested workaround of setting backlight color to #B9EEFF fails when you decrease brightness levels - the backlight ends up looking really blue instead.

My bcdDevice value is 0x04C6 and the OS I'm using is Windows 10.

Thanks!

mathewng, 2020-10-26:
I use fingerprints_two_tone and I found that the more LEDs that are lit up, the worse the colour skew will be.
Red has greater brightness over green and green has greater brightness over blue.

If you are just lighting up a few LEDs, #FFFFFF will appear to be white.

The colour code #B9EEFF is not always accurate.

mathewng, 2020-11-02:
1. Set
`ISSI_Global_Brightness = 35;`
into **Custom KLL**

Tested on my gemini dawn that 35 is the max brightness I can go powering #FFFFFF for all LEDs on the keyboard. 

If your lighting doesn't use all LEDs (#FFFFFF uses all 3 LEDs at full brightness and consumes more power than just #FF0000 or #FFFF00) or colours with less luminance, you can get away with higher brightness values. Your mileage may vary.

2. Plug the second USB port into a USB charger. If there is not enough power, the colour will shift.

If you've done both the above, you will no longer experience colour skew.