kiibohd/controller #389: Kira LEDS go into a continous blinking loop (red, blue, green on different parts of the keyboard)

kajotac, 2020-07-09:
* **Device**: Kira (Kono Store)

* **Firmware Origin**: 0.5.7.11 (github)

* **OS**: Linux (currently arch, kernel 5.6.x)

* **Version**: bcdDevice 0.01

* **Reproduction Steps**: Leave computer unattended for several hours. Screen goes to sleep, but the computer is powered on.

* **Resulting Issue**: After a long period of time the keyboard goes into a continuous blinking loop, showing different colors (red, blue, green) in different parts of the keyboard. The keyboard keeps working, with the exception of the LED animation or light settings. Its important to note that it happens when the system is not being used, the screen goes to sleep, but the system is online. I have a video, but is to big to attach it so I just uploaded some clips in a compressed file.

[drive-download-20200709T200710Z-001.zip](https://github.com/kiibohd/controller/files/4899263/drive-download-20200709T200710Z-001.zip)

haata, 2020-07-09:
This looks like an RF (or possible USB errors) issue.

RF can be caused by a lot of things (like having your cell phone in the perfect spot between your keyboard and the antenna). Unfortunately, it's not possible for the firmware to detect this situation.

Things you can try:
1. Replace the USB cable
2. Try adding an RF choke https://kiibohd.github.io/wiki/#/KiraRF

kajotac, 2020-07-10:
I already tried two different USB cables, but you will be correct about the noise. The Kira is my data center keyboard, and it's pretty darn noisy in there.

I'll have to do the EF choke. I'm sorry I didn't pay attention to the wiki first.

Thank you very much.

kajotac, 2020-07-10:
Could this happen to the Keystone Analog Mechanical Keyboard too?

haata, 2020-07-10:
Unlikely.
1. Led controller is SPI which has better noise immunity than I2C
2. No daughter board for USB connector which is where the noise. So it's much harder for noise to enter.
3. Some additional protection circuits

But I'll be doing some additional testing to make sure. I haven't seen this issue with the SPI led controller.

kajotac, 2020-07-10:
Got it!

Thank you for your response. Item number 2 is an eye opener. I'm building a different board for a friend that uses a daughter board for the USB connector. I might add an RF choke as well, just in case.